stanfordLINQS/SQcircuit #6: This solver cannot solve your circuit.

decod81, 2022-07-08:
Hi. I was hoping to quickly try out different coupling circuits between two transmons and generate numerical Hamiltonians for the system in order to simulate CZ gate fidelities in time domain, but seems SQcircuit is unable to solve most of my circuits. It just gives ValueError: This solver cannot solve your circuit. I give the simplest example of my code here so one can replicate the problem.

My brief investigation suggests some accuracy check fails in circuit.py. If I remove the check, the code executes, but it sees 4 modes instead of 3 as there should be (easy to verify for example in qucat). I guess SQcircuit needs the number of nodes to correspond to the number of modes, but for arbitrary schematics this is often not the case.

I tried using qucat to generate Hamiltonians as well, but qucat only generates near diagonal H and eliminates the coupling terms I need for my simulations so I was hoping SQcircuit could solve my problem. A quick numerical test I did appeared to suggest SQcircuits might be usable the way I hoped for simple capacitively coupled transmons, but maybe not for more complicated circuits. Perhaps also worth mentioning is that to get correct eigenfrequencies even for capacitively coupled qubits, I had to use relatively high truncation number (maybe 10) which resulted in rather large numerical H compared to what would otherwise suffice if one derives it by hand (3) and while perhaps not an issue as such, it would slow down the numerical simulations quite a bit.

I'm not sure if these are bugs or features, but I just felt it might be worthwhile to bring into your attention what I was hoping for from SQcircuit and how I at least so far failed to reach it. Comments and suggestions are appreciated.

<pre>
from numpy import *
import SQcircuit as sqc

J = 2.07e-15/2/pi/1e9/6.6e-34
C = 1.6e-19**2/2/1e9/6.6e-34
L = (2.07e-15/2/pi)**2/1e9/6.6e-34

CJ1 = sqc.Capacitor(C/85e-15)
JJ1 = sqc.Junction(J*32e-9*0.999)
Cc1 = sqc.Capacitor(C/40e-20)

Cr2A = sqc.Capacitor(C/117e-15)
Lr2 = sqc.Inductor(L/2.34e-9)
Cr2B = sqc.Capacitor(C/117e-15)

Cc2 = sqc.Capacitor(C/40e-20)
CJ2 = sqc.Capacitor(C/85e-15)
JJ2 = sqc.Junction(J*32e-9/0.999)

circuit = {(0, 1): [CJ1, JJ1],
           (1, 2): [Cc1],
           (0, 2): [Cr2A],
           (2, 3): [Lr2],
           (0, 3): [Cr2B],
           (3, 4): [Cc2],
           (0, 4): [CJ2, JJ2]}

cir = sqc.Circuit(circuit)
cir.description()
</pre>

taha1373, 2022-07-08:
Hi @decod81, 

Thanks for using SQcircuit and sharing your issue with us.

The error that you encountered is an old piece of code (I should have removed it). Basically, SQcircuit should be able to solve your problem. I will take a look at the circuit and I will get back to you. Probably there is a bug that I should fix.

taha1373, 2022-07-08:
Hi @decod81 

Let me clarify several points regarding your issue. 

* Firstly, the reason the code gave you the error was a bug that I fixed (thanks to you) and it should be resolved in version 0.0.14 that I have already released. Let me know if you have any more issues.
* Second, SQcircuit gives you all the modes, but some of the modes do not participate in the system's dynamic (we called them uncoupled modes in the paper). These modes (either charge or harmonic modes) do not have any flux term in JJ part of the Hamiltonian. For example, in your circuit, the fourth mode is a charge mode and does not have any flux in the JJ part. Therefore, you can omit this mode by setting its truncation number to 1. For example, in your circuit, you should do ``cir.set_trunc_nums([15, 9, 9, 1])``. In future versions, we omit these modes internally and we just give the users modes that are part of the dynamic.
* Third, as you mentioned for two Transmon coupled capacitively, If the coupling is weak, then dividing the system into two separate qubits is a better basis (the way that we analyze these circuits on the paper, and by better basis I mean a basis that needs smaller Hilbert Space to calculate lower eigenfrequencies). However, if the coupling is very strong that you can not partition your system to two Transmon anymore, the current version of SQcircuit is using a better basis. So it is not only about the size of the circuit parameters also matter.  Our current research is focused on how to detect these bases automatically and efficiently. In future versions of SQcircuit, this issue should be improved.

decod81, 2022-07-11:
Thank you very much for the quick fix and clarifications.

Upon my quick testing it seems to work now most of the time. It did fail once though in the middle of a bunch of runs with the same message and then worked again with the next runs (working and failed runs were with identical parameters). I guess there is some random factor that causes fail every now and then, but as long as it doesn't become too frequent in some interesting parameter range, I think it's good.

taha1373, 2022-07-13:
Hi @decod81, 
Thanks again for your great feedback. I noticed the problem. It is an issue with Gram–Schmidt process accuracy. I will fix it in the next patch (0.0.15).