stanfordLINQS/SQcircuit #12: Wrong sign in Transmon Hamiltonian

PhilippAumann, 2022-11-09:
Thank you for providing this software!
When initializing a transmon circuit, I realized that the sign of the potential part of the Hamiltonian, which is printed by the  `description()` function is wrong.
MWE:
```python
import SQcircuit as sq

C = sq.Capacitor(3, 'GHz')
JJ = sq.Junction(10, 'GHz', cap=C)

elements = {(0, 1): [C, JJ]}
circuit = sq.Circuit(elements)
circuit.description()
```

The Hamiltonian is given as `H = EC_11(n_1-ng_1)^2 + EJ_1cos(φ_1)`, while the sign in front of ` EJ_1` should be a minus.
I guess this is just a typo in the way the symbolic Hamiltonian is generated. The numerical representations seems to be fine.

taha1373, 2022-11-13:
Hi @PhilippAumann, 

Thanks for your feedback! You are right it is just a typo bug! I will fix it in 0.0.16-beta patch!

taha1373, 2022-11-14:
@PhilippAumann I just updated  the SQcircuit to v0.0.16-beta. The issue should be solved now.

closed 2022-11-14 by PhilippAumann