stanfordLINQS/SQcircuit #17: Connecting two transmons by an inductor works but not with a capacitor

marekgluza, 2023-01-04:
Hi, coupling two transmons via a capacitor doesn't work but with an inductor it works.

The error message is: 
LinAlgError: Last 2 dimensions of the array must be square

Here an example of what I tried so far
[image]

The code is fun, thanks!

marekgluza, 2023-01-04:
I read a previous issue response @taha1373 in https://github.com/stanfordLINQS/SQcircuit/issues/2#issuecomment-1164886741_
and 'grounding' each transmon indeed works
[image]

taha1373, 2024-08-27:
Hi @marekgluza,

Thank you for using SQcircuit and for taking the time to report the issue. I'm glad that were able to resolve it! If you have any further questions or encounter any other issues, please don't hesitate to reach out.

closed 2024-08-27 by taha1373