stanfordLINQS/SQcircuit #19: The order of the elements in the circuit description matters

GabrielJauma, 2023-08-10:
As you can see in the screenshots, changing the order in which the inductor and capacitor are listed changes the resulting circuit.

[image]
[image]

Anyways, thanks a lot for this code, it is great! Keep it up =)

GabrielJauma, 2023-08-11:
Okay, I think I now understand why this is happening. I believe that I should have introduced both the capacitor and the inductor in a single line of code, that is, as a single entrance to the dictionary. 

Maybe you could include a warning saying that all the elements between two nodes should be included in the same line?

Have a great one =)

taha1373, 2024-08-27:
Thank you, @GabrielJauma, for using SQcircuit and for sharing your issue with us. I appreciate your feedback and will work on adding the requested feature to SQcircuit in the near future. Thanks again for helping us improve!

closed 2024-08-27 by taha1373