stanfordLINQS/SQcircuit #23: Incompatibility with qutip 5

juanjosegarciaripoll, 2024-04-16:
The package expects some `qutip.qobj` module that seems missing. I have only managed to make sqcircuit work by fixing qutip to 4.7.6
```
conda create -n sqcircuit python matplotlib-base
pip install qutip=4.7.6
pip install sqcircuit
```

hodgestar, 2024-04-16:
QuTiP developer here. In the short term while SQcircuit does not yet support QuTiP 5, changing its qutip dependency to specify `qutip~=4.7` seems like a sensible quick fix.

If there are questions regarding porting to QuTiP 5, feel free to ask here, or on the QuTiP mailing list or GitHub issues depending on which seems easiest / most appropriate. We did a fair amount of work to ease migration before the release, and we're keen to help as much as we can.

taha1373, 2024-08-27:
Thank you, @juanjosegarciaripoll, for using SQcircuit and bringing this issue to our attention. I also want to extend my thanks to @hodgestar for providing a quick fix. I'm pleased to inform you that we've released Version 1.0.0 of SQcircuit, which addresses the QuTiP dependency issue. Our latest version now fully supports QuTiP 5.0.0 and above.

closed 2024-08-27 by taha1373

hodgestar, 2024-08-27:
@taha1373 That's awesome! How did the migration to QuTiP 5 go?

sambonkov, 2024-08-29:
@hodgestar Overall moving to QuTiP 5 was mostly smooth. Before we relied on the QuTiP `.data` being stored as a CSR matrix, so it took a little bit to get used to the new options for `data` format and make sure we dealt with that correctly (without calling `.full()` and making it dense).

taha1373, 2024-08-29:
@hodgestar Apologies for the delayed response. I was waiting for @sambonkov's input since he handled the migration.

hodgestar, 2024-08-29:
Glad to hear things went mostly smoothly. Woot.